OpenEtherCATsociety/SOEM #923: Synchronization without DC

kkmserver, 2025-07-23:
I can't get the device to accept and transmit PDO.
As far as I understand, there is no information about DC synchronization in ESI.
Here is an excerpt from ESI file:

.```
..

<Sm MinSize="34" MaxSize="256" DefaultSize="256" StartAddress="#x1000" ControlByte="#x26" Enable="1">MBoxOut</Sm>

<Sm MinSize="34" MaxSize="256" DefaultSize="256" StartAddress="#x1200" ControlByte="#x22" Enable="1">MBoxIn</Sm>

<Sm MinSize="1" MaxSize="256" DefaultSize="6" StartAddress="#x1300" ControlByte="#x24" Enable="1">Outputs</Sm>

<Sm MinSize="1" MaxSize="256" DefaultSize="6" StartAddress="#x1600" ControlByte="#x20" Enable="1">Inputs</Sm>
....
<Mailbox DataLinkLayer="true">
	<CoE SdoInfo="false" PdoConfig="true" PdoAssign="true" CompleteAccess="false" **SegmentedSdo="false"**/>
</Mailbox>
```

There is no <Dc> tag in ESI file.

In Op-state comes out without problems.
But when trying to transmit PDO, the "Time-Out" LED lights up on the device (what this means is not described in the documentation)

Do I need to somehow configure synchronization if it is without DC?

And can you tell me what <SegmentedSdo="false"> means in the <Mailbox> <CoE> tag???

Also, one of the oddities is SDO with a null index in PDO:
```
<TxPdo Fixed="0" Mandatory="1" Sm="3">
	<Index>#x1A00</Index>
	<Name>1st TxPDO Mapping</Name>
	<Entry>
		<Index>#x6041</Index>
		<SubIndex>0</SubIndex>
		<BitLen>16</BitLen>
		<Name>Status Word</Name>
		<DataType>UINT</DataType>
	</Entry>
	<Entry>
		<Index>#x6043</Index>
		<SubIndex>0</SubIndex>
		<BitLen>16</BitLen>
		<Name>vl velocity demand</Name>
		<DataType>INT</DataType>
	</Entry>
	<Entry>
		<Index>#x6061</Index>
		<SubIndex>0</SubIndex>
		<BitLen>8</BitLen>
		<Name>Mode Of Operation Display</Name>
		<DataType>SINT</DataType>
	</Entry>
	<Entry>
		**<Index>#x0000</Index>**
		<SubIndex>0</SubIndex>
		<BitLen>8</BitLen>
	</Entry>
</TxPdo>
```

Thanks

nakarlsson, 2025-07-25:
Post a Wireshark when the issue occur when in OP. Also, add a printout of the slaveinfo

kkmserver, 2025-07-26:
The device turned out to have a faulty EEPROM:

Slave:1
 Name:MS300(CMM-EC02 Card)
 Output size: 32bits
 Input size: 40bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 000001dd ID: 10400340 Rev: 00020000
 SM0 A:1000 L: 256 F:00010026 Type:1
 SM1 A:1200 L: 256 F:00010022 Type:2
 SM2 A:1300 L:   5 F:00010024 Type:4
 SM3 A:1600 L:   4 F:00010020 Type:3
 FMMU0 Ls:00000000 Ll:   4 Lsb:0 Leb:7 Ps:1600 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000004 Ll:   5 Lsb:0 Leb:7 Ps:1300 Psb:0 Ty:01 Act:01
 FMMU2 Ls:00000009 Ll:   1 Lsb:0 Leb:7 Ps:080d Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 256 rd: 256 MBX protocols : 04
 CoE details: 0d FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 inputs
     addr b   index: sub bitl data_type    name
**Segmentation fault**

Sorry again.
Closing...

closed 2025-07-26 by kkmserver