OpenEtherCATsociety/SOEM #924: Use SOEM for PDO Mapping

Kwahi11, 2025-07-24:
I am now using soem to configure PDO to control the motor, but I encountered the following problems when configuring the CiA 402 standard PDO:
1. When I manually configure PDO, 0x1c12 and 0x1c13 can be configured normally, but 0x1600 and 0x1a00 fail to write (wkc = 0).

**My program** (based on soem's simpleTest to change pdo and encapsulate ec_write) running results;
`:~/SOEM/test/linux/simple_test/build$ sudo ./Mytest enp1s0
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on enp1s0 succeeded.
1 slaves found and configured.
Slave 1 has CA? true
Slave: 1 - Write at 0x1c12:0 => wkc: 1; data: 0x0	{0x1C12 0 = 0}
Slave: 1 - Write at 0x1600:0 => wkc: 0; data: 0x0	{0x1600 0 = 0}
Slave: 1 - Write at 0x1600:1 => wkc: 0; data: 0x60400010	{0x1600 1 = 0x60400010}
Slave: 1 - Write at 0x1600:2 => wkc: 0; data: 0x607a0020	{0x1600 2 = 0x607A0020}
Slave: 1 - Write at 0x1600:3 => wkc: 0; data: 0x60ff0020	{0x1600 3 = 0x60FF0020}
Slave: 1 - Write at 0x1600:4 => wkc: 0; data: 0x60710010	{0x1600 4 = 0x60710010}
Slave: 1 - Write at 0x1600:5 => wkc: 0; data: 0x60720010	{0x1600 5 = 0x60720010}
Slave: 1 - Write at 0x1600:6 => wkc: 0; data: 0x60600008	{0x1600 4 = 0x60600008}
Slave: 1 - Write at 0x1600:7 => wkc: 0; data: 0x0008	{0x1600 4 = 0x00000008}
Slave: 1 - Write at 0x1600:0 => wkc: 0; data: 0x7	{0x1600 0 = 7}
Slave: 1 - Write at 0x1c12:1 => wkc: 1; data: 0x1600	{0x1C12 1 = 0x1600}
Slave: 1 - Write at 0x1c12:0 => wkc: 1; data: 0x1	{0x1C12 0 = 1}
Slave: 1 - Write at 0x1c13:0 => wkc: 1; data: 0x0	{0x1C13 0 = 0}
Slave: 1 - Write at 0x1a00:0 => wkc: 0; data: 0x0	{0x1A00 0 = 0}
Slave: 1 - Write at 0x1a00:1 => wkc: 0; data: 0x60410010	{0x1A00 1 = 0x60410010}
Slave: 1 - Write at 0x1a00:2 => wkc: 0; data: 0x60640020	{0x1A00 2 = 0x60640020}
Slave: 1 - Write at 0x1a00:3 => wkc: 0; data: 0x606c0020	{0x1A00 3 = 0x606C0020}
Slave: 1 - Write at 0x1a00:4 => wkc: 0; data: 0x60770010	{0x1A00 4 = 0x60770010}
Slave: 1 - Write at 0x1a00:5 => wkc: 0; data: 0x603f0010	{0x1A00 5 = 0x603F0010}
Slave: 1 - Write at 0x1a00:6 => wkc: 0; data: 0x60610008	{0x1A00 6 = 0x60610008}
Slave: 1 - Write at 0x1a00:7 => wkc: 0; data: 0x0008	{0x1A00 7 = 0x00000008}
Slave: 1 - Write at 0x1a00:0 => wkc: 0; data: 0x7	{0x1A00 0 = 7}
Slave: 1 - Write at 0x1c13:1 => wkc: 1; data: 0x1a00	{0x1C13 1 = 0x1a00}
Slave: 1 - Write at 0x1c13:0 => wkc: 1; data: 0x1	{0x1C13 0 = 1}
Slave: 1 - Write at 0x6060:0 => wkc: 1; data: 0x9	{Set OpMode = 9 (CSV)}
Slave: 1 - Read at 0x6061:0 => wkc: 1; data: 0x0 (0)	[OpMode Display]
Slave: 1 - Write at 0x60ff:0 => wkc: 1; data: 0x0000	{Target velocity = 0}
Slave: 1 - Write at 0x6040:0 => wkc: 1; data: 0x06	{Controlword = Shutdown}
Slave: 1 - Read at 0x6041:0 => wkc: 1; data: 0x1210 (4624)	[Statusword]
Slave: 1 - Write at 0x6040:0 => wkc: 1; data: 0x07	{Controlword = Switch On}
Slave: 1 - Read at 0x6041:0 => wkc: 1; data: 0x1210 (4624)	[Statusword]
Slave: 1 - Write at 0x6040:0 => wkc: 1; data: 0x0f	{Controlword = Enable Op}
Slave: 1 - Read at 0x6041:0 => wkc: 1; data: 0x1210 (4624)	[Statusword]
Slave: 1 - Write at 0x60ff:0 => wkc: 1; data: 0x03e8	{Target velocity = 1000}
Slave: 1 - Write at 0x6060:0 => wkc: 1; data: 0x3	{OpMode}
Slave: 1 - Read at 0x6061:0 => wkc: 1; data: 0x0 (0)	[OpMode display]

Slave:1
 Name:MT_Device
 Output size: 128bits
 Input size: 128bits
 State: 2
 Delay: 0[ns]
 Has DC: 1
 ALcode:0
Slaves mapped, state to SAFE_OP.
ret ec_statecheck = 4
segments : 1 : 32 0 0 0
Request operational state for all slaves
Calculated workcounter 3
1 slave state: 4
Operational state reached for all slaves.
1 slave state: 4
Not all slaves reached operational state.
Slave 1 State=0x04 StatusCode=0x0000 : No error

Request init state for all slaves
Slave: 1 - Write at 0x10f1:2 => wkc: 1; data: 0x0000	{Heartbeat}
End simple test, close socket
End program
`

**My running results using soem's simple_test.c**:
`:~/下载/SOEM-1.3.3-beta.1/build/test/linux/simple_test$ sudo ./simple_test enp1s0
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on enp1s0 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 32 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Not all slaves reached operational state.
Slave 1 State=0x04 StatusCode=0x0000 : No error

Request init state for all slaves
End simple test, close socket
End program
`

**My running results of slaveinfo.c**:
`~/下载/SOEM-1.3.3-beta.1/build/test/linux/slaveinfo$ sudo ./slaveinfo enp1s0
[sudo] xxx 的密码： 
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enp1s0 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:MT_Device
 Output size: 128bits
 Input size: 128bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00202008 ID: 00000000 Rev: 00010000
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:  16 F:00010064 Type:4
 SM3 A:1400 L:  16 F:00010020 Type:3
 FMMU0 Ls:00000000 Ll:  16 Lsb:0 Leb:7 Ps:1400 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000010 Ll:  16 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 27 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program
`

**### What I have tried:**
Change the location of enabling 0x1c12 and 0x1c13, but enabling 0x1c12 and 0x1c13 can be written anywhere, but 0x1600 and 0x1a00 still fail to write.

Now I can't configure PDO, so I can't debug the motor. I hope I can get help here

nakarlsson, 2025-07-25:
Post a Wireshark when the issue occur

Kwahi11, 2025-07-30:
Hello, I'll attach the wireshark results from running simpletest.c, 

[MT-Device_xml.txt](https://github.com/user-attachments/files/21500138/MT-Device_xml.txt)

[result01.txt](https://github.com/user-attachments/files/21500148/result01.txt)

as well as the official ESI file. While reviewing the ESI file recently, I discovered a few issues:

1. The ESI file defines 1600 in 0x1c12 and 1a00 in 1c13 as read-only. This likely explains my previous PDO configuration failure, meaning I only need to configure 1c12 and 1c13 for this motor (I suspect).
2. If I only need to configure 1c12 and 1c13, I've already done that, but it's stuck in the safe-op state. I previously thought this was due to a PDO configuration failure, but it seems that's not the case. I can't find the cause.
3. I've read this link: https://github.com/OpenEtherCATsociety/SOEM/issues/541, but its ESI file contains information about Initcmd, while my ESI doesn't, so I'm still at a loss.

Kwahi11, 2025-08-01:
### [simpleTest.pcapng.txt](https://github.com/user-attachments/files/21541146/simpleTest.pcapng.txt)

Hello, the above is the original Wireshark for my SimpleTest running SOEM as it is. Just delete the suffix. txt. The following is the result of the operation,

`xxx@xxx:~/ethercat/SOEM-1.3.3-beta.1/build/test/linux/simple_test$ sudo ./simple_test enp1s0
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on enp1s0 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 32 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Not all slaves reached operational state.
Slave 1 State=0x04 StatusCode=0x0000 : No error

Request init state for all slaves
End simple test, close socket
End program
  `
According to the running result, the AL status code is 0x0000 (no error), but it is still stuck in the safeop state. Regarding other information, you can refer to my first reply, where I ran my own test file (mainly changed PDO configuration and DC, and added printing all process information).
thanks in advance

Kwahi11, 2025-08-01:
[simpleTestResult.pcapng.txt](https://github.com/user-attachments/files/21541301/simpleTestResult.pcapng.txt)

Hello, the above is the original Wireshark for my SimpleTest running SOEM as it is. Just delete the suffix. txt. The following is the result of the operation,

`xxx@xxx:~/ethercat/SOEM-1.3.3-beta.1/build/test/linux/simple_test$ sudo ./simple_test enp1s0
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on enp1s0 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 32 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Not all slaves reached operational state.
Slave 1 State=0x04 StatusCode=0x0000 : No error

Request init state for all slaves
End simple test, close socket
End program
`
According to the running result, the AL status code is 0x0000 (no error), but it is still stuck in the safeop state. Regarding other information, you can refer to my first reply, where I ran my own test file (mainly changed PDO configuration and DC, and added printing all process information).
thanks in advance

Kwahi11, 2025-08-01:
1. I thought it had previously reported ALstatuescode = 0x001b, indicating a watchdog error, so I tried disabling the watchdog by writing zero to index 1 of 0xF1, but it still had no effect.

2. One issue is that according to the Wireshark datagram, the master initially sends 11 (init+op) to the slave at ADP:0X1, ADO:0X120. ChatGPT says this is abnormal, but I'm not sure.

Kwahi11, 2025-10-30:
> I ran into the same issue. How did you end up solving it? Maybe we can discuss it together?

I eventually discovered it was due to an incorrect motor hardware version. I had the motor manufacturer update the firmware, and then it worked normally. My guess is that your motor slave's software or hardware version might be too old. I've found that SOEM doesn't have very good compatibility with older versions, so I think you should first confirm that the hardware is fine. If you still have problems, feel free to contact me.

closed 2025-10-30 by Kwahi11

CactuserLi, 2025-11-03:
Hi, I ran into the same issue. But my POD mappings are all read-only.Do you have any clues?
Looking forward to your reply

yishoucheng, 2025-11-03:
Hello senior, I am using TWINCAT3 as the master station. When porting the slave station, I encountered the following problem during the transition from PRE-OP to SAFE-OP state: timeout downloading PDO 0x1c12 index. Have you ever encountered a similar issue? If so, I would greatly appreciate your guidance.